flipperzero-rs/flipperzero-rs #15: Supported API version?

python273, 2022-11-21:
Do I understand correctly 7.5 is not supported?

dcoles, 2022-11-21:
That's correct. I have a WIP branch ([`sdk7.x`](https://github.com/dcoles/flipperzero-rs/tree/sdk7.x)), which brings everything up to date; but the SDK version check fails on newer firmware and I'm still trying to debug that.

python273, 2022-11-21:
Cool. I manually updated `API_VERSION` to match header from another compiled app (should be `7.51` from [unleashed](https://github.com/DarkFlippers/unleashed-firmware)), it passes version check, but the app stops immediately
```diff
diff --git a/crates/sys/src/bindings.rs b/crates/sys/src/bindings.rs
index a38fb8b..e4c3087 100644
--- a/crates/sys/src/bindings.rs
+++ b/crates/sys/src/bindings.rs
@@ -110,7 +110,7 @@ impl<T> ::core::fmt::Debug for __IncompleteArrayField<T> {
         fmt.write_str("__IncompleteArrayField")
     }
 }
-pub const API_VERSION: u32 = 458752;
+pub const API_VERSION: u32 = 3342343;
 #[repr(C)]
 #[derive(Debug, Copy, Clone)]
 pub struct Bt {
```

python273, 2022-11-21:
I guess major and minor should be in the reverse order
https://github.com/dcoles/flipperzero-rs/blob/57df18579b92b9340103a4958676291657a07c8d/tools/src/bin/generate-bindings.rs#L58

python273, 2022-11-21:
> app stops immediately

oh, that's what it supposed to do 😅 figured out how to see logs via serial, prints `Hello, Rust!`, so seems to be working

dcoles, 2022-11-21:
> I guess major and minor should be in the reverse order

Ah, thank you! I forgot that endianess causes all sorts of havoc when dealing with type punning.
If that fixes everything, I should be able to tag a new version shortly.

dcoles, 2022-11-23:
Released flipperzero-rs v0.4.0 with support for SDK 7.5 (see [crates.io](https://crates.io/crates/flipperzero/0.4.0)).

Let me know if you have any issue.

closed 2022-11-23 by dcoles