flipperzero-rs/flipperzero-rs #25: Failed to build hello-rust example

valentinegb, 2022-12-22:
When trying to build the `hello-rust` example (`cargo build --release`), it fails and prints the following error:
```
error: the 'cargo' binary, normally provided by the 'cargo' component, is not applicable to the 'nightly-2022-11-03-aarch64-apple-darwin' toolchain
```

## Steps to reproduce
 1. Download ZIP from GitHub
 2. Move `hello-rust` folder out (I moved it to my `projects` folder)
 3. In the `hello-rust` directory, run the following commands:
     1. `rustup default nightly`
     2. `rustup target add thumbv7em-none-eabihf`
     3. `cargo build --release`
 4. Build fails

## Desktop
 - **OS:** macOS Ventura 13.2 (a) Beta (22D7750270d)
 - **`rustup`:** 1.25.1 (bb60b1e89 2022-07-12)
 - **`rustc`:** 1.67.0-nightly (edf018221 2022-11-02)
 - **`cargo`:** 1.67.0-nightly (7e484fc1a 2022-10-27)

katekyy, 2022-12-26:
Try uninstalling your toolchain with `rustup uninstall nightly-2022-11-03-aarch64-apple-darwin`, and installing it again.

valentinegb, 2022-12-26:
Alright, seems like that worked!

```
valentinebriese@Valentines-Air hello-rust % rustup uninstall nightly-2022-11-03-aarch64-apple-darwin
info: uninstalling toolchain 'nightly-2022-11-03-aarch64-apple-darwin'
info: toolchain 'nightly-2022-11-03-aarch64-apple-darwin' uninstalled
valentinebriese@Valentines-Air hello-rust % cargo build --release
info: syncing channel updates for 'nightly-2022-11-03-aarch64-apple-darwin'
info: latest update on 2022-11-03, rust version 1.67.0-nightly (edf018221 2022-11-02)
info: downloading component 'cargo'
info: downloading component 'clippy'
info: downloading component 'rust-docs'
info: downloading component 'rust-std'
info: downloading component 'rust-std' for 'thumbv7em-none-eabihf'
info: downloading component 'rustc'
info: downloading component 'rustfmt'
info: installing component 'cargo'
info: installing component 'clippy'
info: installing component 'rust-docs'
info: installing component 'rust-std'
info: installing component 'rust-std' for 'thumbv7em-none-eabihf'
info: installing component 'rustc'
info: installing component 'rustfmt'
    Updating crates.io index
   Compiling flipperzero-sys v0.5.0
   Compiling flipperzero-rt v0.5.0
   Compiling flipperzero v0.5.0
   Compiling hello-rust v0.1.0 (/Users/valentinebriese/projects/hello-rust)
    Finished release [optimized] target(s) in 7.41s
valentinebriese@Valentines-Air hello-rust % cargo build --release
    Finished release [optimized] target(s) in 0.01s
valentinebriese@Valentines-Air hello-rust % cargo --version
cargo 1.67.0-nightly (7e484fc1a 2022-10-27)
```

Dunno why that happened, but I guess I'll close the issue then.

closed 2022-12-26 by valentinegb

dcoles, 2022-12-30:
Thanks @katekyy!

I'm also at a bit of a loss why that might have happened, but glad to hear you got it resolved.