flipperzero-rs/flipperzero-rs #31: Notification_messages example?

ASoldo, 2023-01-08:
Hi, can we get simple example how to use notification/notification_messages in rust? So far demo with DialogMessageButtons is only way we can interact with key presses. I am new to Rust and Flipper Zero and would really appropriate sample how to achieve this. Simple key input check and log message.
Thank you!

dcoles, 2023-01-09:
Hi @ASoldo,

Nice to hear that you're using the Flipper Zero as a way of learning Rust!

Sadly the [`flipperzero`](https://crates.io/crates/flipperzero) crate doesn't currently expose GPIO or notifications, but it's not too painful to do so using the low-level [`flipperzero-sys`](https://crates.io/crates/flipperzero-sys) crate.

I've just added a [notification example](https://github.com/flipperzero-rs/flipperzero/blob/main/examples/notification) showing how this can be done. I still need to take a look at what GPIO would require.

ASoldo, 2023-01-09:
That is great, thank you. If you could take a look at GPIO and simple toggle open pin/close pin, I will have more than enough to start playing. Since I do have all that in C. But I kinda like Rust and would like to switch to it 100%. This would help me a lot. Also, I can share demos faster and document stuff myself.

dcoles, 2023-01-09:
I dug out the breadboard and was able to get a [GPIO example](https://github.com/flipperzero-rs/flipperzero/tree/main/examples/gpio) working. 

[image]

The biggest issue I found is that a number of the [`furi_hal_gpio_`](https://github.com/flipperdevices/flipperzero-firmware/blob/0.75.0/firmware/targets/f7/furi_hal/furi_hal_gpio.h) functions are defined `inline`, thus there's no implementation in the SDK. Hence the need for reimplementing `gpio_write`.

Let me know if that's enough to get you unblocked. Happy hacking!

ASoldo, 2023-01-09:
Thank you for your quick response, I do appreciate time and effort. I will try it right now. I have one question rn, since you mention that gpio_write needs re implementation, is that something that will take some time to do or that could be done fairly easy? Just to know for future reference when I get back for more examples, will this example work or I will need to use exposed api.

closed 2023-01-09 by dcoles