flipperzero-rs/flipperzero-rs #43: Add ability to flash / launch a fap from Rust

twitchyliquid64, 2023-02-22:
It would be good to remove the remaining python/fbt tooling from the development toolchain/flow entirely. I had a quick looksies at how the official flipper zero tooling does file transfers and it looks like it might actually be very easy:

https://github.com/flipperdevices/flipperzero-firmware/blob/663eb6cd6d4a14d00caf93d402b71eadb9a4d63f/scripts/flipper/storage.py#L203

I'm getting back into Flipper stuff again (needless to say I need to rewrite all my trash onto your mature crates lol), so I might give this a go if you don't get around to it first.

dcoles, 2023-02-23:
Hi @twitchyliquid64,

That would be great! That was part of the intent for the [`tools`](https://github.com/flipperzero-rs/flipperzero/tree/main/tools) crate (a simple serial client, storage client, etc.), but hadn't had the opportunity.

Would happily accept something like that.

dcoles, 2023-03-06:
I've started working on serial support in #54.

twitchyliquid64, 2023-03-08:
I had attempted to do it in bash but gave up once i saw how easy it was to copypasta `storage.py` and call one method.

The draft looks great!

dcoles, 2023-03-12:
Merged in support for [`serial_cli`](https://github.com/flipperzero-rs/flipperzero/tree/01ae5bc1969a66ed0f9f0667572df671fec5c57b/tools#serial_cli) and [`storage`](https://github.com/flipperzero-rs/flipperzero/tree/01ae5bc1969a66ed0f9f0667572df671fec5c57b/tools#storage) tools. 🎉 

Would greatly appreciate if you could check it also works on your platform.

closed 2023-03-12 by dcoles

twitchyliquid64, 2023-03-13:
I was able to run the serial_cli tool, but the `CTRL+]` to exit thing didnt work for me.