flipperzero-rs/flipperzero-rs #53: Test harness

str4d, 2023-03-05:
Currently all testing facilities are disabled for the crates in this repo. It makes sense because everything will only run on an actual Flipper Zero, but it does make verification of things tricky.

It would be useful to have a test harness, that runs the test binaries on a connected Flipper Zero. I'm imagining something like the [integration tests for the `yubikey` crate](https://github.com/iqlusioninc/yubikey.rs/blob/main/tests/integration.rs), which require setting a `YUBIKEY_SERIAL` environment variable in order to run. The [`test-with`](https://crates.io/crates/test-with) crate might also be helpful here, specifically `#[test_with::env(FLIPPER_ZERO_SERIAL_PORT)]` or something, that can be used to enable at compilation time tests that require a connected Flipper Zero.

str4d, 2023-03-05:
Ideally the test harness would make it easy to write normal Rust tests that Just Work.™️ What I'm imagining is something like this:
- We write normal tests, but maybe annotated with `#[device_test]` or something to trigger the harness.
  - Alternatively, we do everything inside an integration test suite (though that prevents testing internal APIs).
- The test harness builds a test binary as a Flipper App, that:
  - Runs each test, capturing all panics.
  - Reports the result of each test to the computer over the serial connection.
- The test harness deploys the test app to the specified Flipper Zero and launches it.
  - This step may block on #43.
- The tests run, and results are collected.
- Once the test is complete, the test harness removes the test app from the Flipper Zero.

dcoles, 2023-03-05:
Relevant: [`flipperzero-firmware` Unit Tests documentation](https://github.com/flipperdevices/flipperzero-firmware/blob/0.78.1/documentation/UnitTests.md)

dcoles, 2023-03-05:
I also have a ST-Link USB dongle around somewhere which could be used to catch crashes with the in-circuit debugging feature.

JarvisCraft, 2023-03-06:
It may also be worth looking at something like QEMU (as mentioned in [The Embedded Rust Book]), at least for running some tests, at least the ones which don't access periphery (screen, GPIO, iButton, etc). This would probably allow running these tests on CI.

[The Embedded Rust Book]: https://docs.rust-embedded.org/book/start/qemu.html

JarvisCraft, 2023-03-06:
Also to add to the list os useful links:
- [*Testing an embedded application*](https://ferrous-systems.com/blog/test-embedded-app/)

str4d, 2023-03-19:
Okay, #57 now implements on-device unit and integration tests with the kind of usability I was aiming for.

dcoles, 2023-03-25:
I'm traveling overseas this week, but should be able to take a look in the evening.

PR #57 (2023-04-25): Test harness
Closes #53.
closed 2023-04-25 by dcoles