flipperzero-rs/flipperzero-rs #64: `rt`: Add framework for creating statics

str4d, 2023-05-10:
Normal Rust binaries will initialize statics on start, and then essentially leak the memory. This works fine because the memory is cleaned up during process exit. We can't assume the same as a FAP, so if we want to support "statics", we should provide a way in the runtime for initializing them before the user's app code runs, and freeing them after it exits.

dcoles, 2023-05-10:
In this case you're talking about statics that are stored on the heap (from [lazy_static](https://crates.io/crates/lazy_static), [once_cell](https://crates.io/crates/once_cell)), rather than static data stored in the `.rodata` right?

str4d, 2023-05-10:
Yep. People could still use one of those crates, but I believe it would get reported as a memory leak by the FZ tooling.

dcoles, 2023-05-27:
It might be possible to provide a factory function for [`OnceCell`](https://docs.rs/once_cell/1.17.1/once_cell/sync/struct.OnceCell.html) that registers the static for cleanup. Calling the [`take`](https://docs.rs/once_cell/1.17.1/once_cell/sync/struct.OnceCell.html#method.take) method takes ownership of the inner value allowing them to be dropped.

str4d, 2023-05-28:
We can't use `once_cell` because it requires `std` for `Mutex`. I'm instead planning on using `generic_once_cell`, which works for any mutex that implements `lock_api::RawMutex`. I've implemented the latter locally, and opened #74 with an `Instant` impl that I need.

dcoles, 2023-05-29:
> We can't use `once_cell` because it requires `std` for `Mutex`.

I think it's possible to use with `#[no-std]` as long as you provide a [`critical-section`](https://crates.io/crates/critical-section) implementation. I might take a look at that today, since a number of embedded crates are based around it.