flipperzero-rs/flipperzero-rs #89: Add support for Fast FAP loading

str4d, 2023-06-21:
https://github.com/flipperdevices/flipperzero-firmware/pull/2790 bumps the SDK to 31 with 15x faster loading of FAP files. It requires adding `.fast.rel` sections, so we will need to alter `flipperzero-rt` to enable it.

dcoles, 2023-06-22:
Looking at the PR, I think what it's doing is creating a hash table for symbol lookup, rather than a linear search.

This might need to be done as some sort of post-processing step as I don't think it's possible to do this in `build.rs` or the [linker script](https://github.com/flipperzero-rs/flipperzero/blob/main/crates/rt/flipperzero-rt.ld). This appears to be what [`scripts/fastfap.py`](https://github.com/flipperdevices/flipperzero-firmware/pull/2790/files#diff-a0e50a592b3e018edf7898a32074ccd786babb9553d3099ac7920c24b8ef983e) is for.

DrZlo13, 2023-06-22:
Yep, [fastfap.py](https://github.com/flipperdevices/flipperzero-firmware/pull/2790/files#diff-a0e50a592b3e018edf7898a32074ccd786babb9553d3099ac7920c24b8ef983e) is post-processing elf files not only for hashing, but also to create an optimized relocation structure for our elf loader. This removes a large number of calls to the storage, and that's how such an acceleration is achieved.

I recommend you to wait until this feature is released because the format of these sections is subject to change. I also left a fallback to the old usual linking process if optimized sections is not present.

dcoles, 2023-06-22:
@DrZlo13 Thanks for the information. Definitely appreciate the fallback, so this isn't an urgent thing to address.

str4d, 2023-08-09:
This was released in firmware version 0.86.1, and once #93 merges we will require at least 0.87.0, so we can start working on this.

str4d, 2023-08-26:
This is now implemented in #99. I _think_ I got the encodings right, and it appears to work (`cargo test` gets noticeably faster) but I'm unsure if the MD5 for tempfile names in `fastfap.py` is just for convenience or meant to be load-bearing.

DrZlo13, 2023-08-27:
@str4d MD5 for temporary filenames is only for reducing filename size. Not all filesystems can handle files named from a C++ generated section name.

str4d, 2023-08-27:
@DrZlo13 thanks, that clarifies things (I didn't know `objcopy` derives section names from filenames in this way).

I expect that means my current PR is adopting section names from the random file name being chosen by [`NamedTempFile`](https://docs.rs/tempfile/3.8.0/tempfile/struct.NamedTempFile.html), which defaults to `.tmp<6_RANDOM_CHARS>`. Making it deterministic on the section name seems like a better approach.

closed 2023-08-28 by str4d