flipperzero-rs/flipperzero-rs #133: Creating new app with template gives error after deploy to FZ

ASoldo, 2024-03-16:
Hi, after updating FZ to latest development branch and created demo app i get this error (that ApiTooOld). Anybody know what to do here?
[image]

JarvisCraft, 2024-03-17:
@ASoldo, hi there!
I've just opened the PR to update to the lastest SDK version: #134.

Please try using its branch and test if it fixes the issue.

ASoldo, 2024-03-17:
This is what I did but I get the same error message after moving .fap to FZ and running it.
```toml
[dependencies]
flipperzero = { git = "https://github.com/flipperzero-rs/flipperzero", branch = "sdk-0.99.1" }
flipperzero-sys = { version = "0.11.0" }
flipperzero-rt = { version = "0.11.0" }
```

JarvisCraft, 2024-03-17:
You should also update `-sys` and `rt` references to the same repo

ASoldo, 2024-03-17:
I did try that but it gives me this error when `cargo build`
```toml
[dependencies]
flipperzero = { git = "https://github.com/flipperzero-rs/flipperzero", branch = "sdk-0.99.1" }
flipperzero-sys = { git = "https://github.com/flipperzero-rs/flipperzero", branch = "sdk-0.99.1" }
flipperzero-rt = { git = "https://github.com/flipperzero-rs/flipperzero", branch = "sdk-0.99.1" }
```
[image]

JarvisCraft, 2024-03-17:
Hmm, that's strange; I will take a look at it

JarvisCraft, 2024-03-18:
Oh, I get it now.

We haven't updated the template repository yet, but the signature of the main function had changed recently. Now it accepts an `Option<CStr>` instead of the raw `*const u8`.

You just have to update it in your application as done in #125.

JarvisCraft, 2024-03-18:
Also, the above PR with API version bump has been merged so you can now use `main` branch.

ASoldo, 2024-03-18:
Yes, now it works, I have switched to `main` and manually update main function.

ASoldo, 2024-03-18:
Yes, now it works, I have switched to main and manually update main function.

ASoldo, 2024-03-21:
After today's FZ firmware update I face the same (ApiTooOld) after successful build and run on device.

dcoles, 2024-03-24:
Hi @ASoldo,

Any time there is a major revision to the Flipper API, we gave to regenerate the bindings since the Flipper API is only compatible across minor API changes. Unfortunately this happens every few firmware updates (any ABI breakage triggers a new major SDK version).

One option is to [manually regenerate the bindings](https://github.com/flipperzero-rs/flipperzero/blob/main/docs/updating-sdk.md). Typically the actual changes are pretty minor, so don't require any changes on the Rust side. The other is just [sticking on the last supported SDK/firmware](https://github.com/flipperzero-rs/flipperzero/blob/main/README.md#sdk-version).

We have #41 as an issue to automate the generation of the bindings. I'd personally like to move away from bundling the bindings in the repository and have them generated at Rust build time, but we might also be able to add some better automation so someone doesn't have to manually check-in an updated version of the bindings.

I'm going to close this issue for the moment, since this will keep happening any time there's a major SDK version jump. However I'll see if I can come up with a better solution going forward.

closed 2024-03-24 by dcoles