flipperzero-rs/flipperzero-rs #141: Bump API Support

Coded-Alchemy, 2024-04-15:
this seems not to support the latest.

Coded-Alchemy, 2024-04-15:
The template needs updating as well, please and thank you.

JarvisCraft, 2024-05-01:
I've proposed a PR with the update to the bindings version.

Coded-Alchemy, 2024-05-01:
> I've proposed a PR with the update to the bindings version.

I tried to test but I dont know how to get around the [template compatibility](https://github.com/flipperzero-rs/flipperzero-template?tab=readme-ov-file#flipperzero-template). Is there anything different I should be doing? Thanks for your efforts on this, its much appreciated!

Coded-Alchemy, 2024-05-01:
> I've proposed a PR with the update to the bindings version.

First up, the efforts on this is greatly appreciated!

Unless Ive done something wrong, this is the error from the sync branch in the template repo that I get running cargo build:
`error[E0412]: cannot find type `CStr` in this scope
  --> src/main.rs:26:24
   |
26 | fn main(_args: Option<&CStr>) -> i32 {
   |                        ^^^^ not found in this scope
   |
help: consider importing this struct
   |
10 + use core::ffi::CStr;
   |

error[E0308]: mismatched types
  --> src/main.rs:23:1
   |
23 | entry!(main);
   | ^^^^^^^^^^^^
   | |
   | expected fn pointer, found fn item
   | expected due to this
   |
   = note: expected fn pointer `fn(*mut u8) -> _`
                 found fn item `for<'a> fn(Option<&'a {type error}>) -> _ {main}`
   = note: this error originates in the macro `entry` (in Nightly builds, run with -Z macro-backtrace for more info)

Some errors have detailed explanations: E0308, E0412.`

JarvisCraft, 2024-05-02:
@Coded-Alchemy, the fix to the template has been merged into `master`.

JarvisCraft, 2024-05-02:
@Coded-Alchemy, looks like I haven't imported the `CStr` (my bad). You can just `use` it as Cargo suggests (I will further fix the template to include the import).

closed 2024-05-03 by JarvisCraft