flipperzero-rs/flipperzero-rs #150: Error: Outdated App - SDK Upgrade

Maddox-Werts, 2024-08-10:
### Hello, Everyone!

I've just updated my Flipper Zero to the latest version of Momentum Firmware. I've also gone ahead and setup a Docker environment to build the application and get everything in order to create the *.FAP* file.

After some Trial and Error, I got Rust to compile to a *.FAP* file. I placed the file on my Flipper Zero's *SD Card* and launched it to get the following message:

[image]

Does anyone know how to upgrade the SDK version? Any help would be greatly appreciated.

JarvisCraft, 2024-08-19:
Hi there and sorry for the late reply!

We are working on update to the latest API in #154 (regenerated bindings) and #155 (additional changes needed in our runtime implementation).

bkstephen, 2024-09-06:
Any update on this?

Maddox-Werts, 2024-09-06:
Hey, Bkstephen! 
No, unfortunately not. I was not able to compile it to an updated version of the SDK. I had to write my app in C. I'd recommend you do the same even though Rust is amazing.

bkstephen, 2024-09-06:
Ok thank you, maybe its worth archiving this project then.

str4d, 2024-09-10:
No this project will not get archived. We just haven't been able to keep up with the very rapid rate of changes to the Flipper Zero SDK prior to its 1.0 release, so we only generally got updates out that jumped over multiple SDK updates.

[Firmware 1.0 was released today](https://blog.flipper.net/released-firmware-1/), so we will target that for the next update.

ASoldo, 2024-09-11:
This is what I've been waiting for. No rush, just make it happen eventually. I can't wait to get back to it again!

8de2fdb0, 2024-09-12:
@str4d 

if you can't wait use
```toml
[dependencies]
flipperzero = { git = "https://github.com/flipperzero-rs/flipperzero", branch = "update-bindings-1.0" }
flipperzero-sys = {  git = "https://github.com/flipperzero-rs/flipperzero", branch = "update-bindings-1.0"  }
flipperzero-rt = {  git = "https://github.com/flipperzero-rs/flipperzero", branch = "update-bindings-1.0"  }
```

@ASoldo nice work

closed 2024-09-12 by dcoles

dcoles, 2024-09-12:
The [`flipperzero-0.12.0`](https://crates.io/crates/flipperzero/0.12.0) crates supporting SDK 73 (matching Firmware 1.0.1) has been released. Thank you for your patience while we made these updates.

PR #157 (2024-09-12): Update SDK bindings to 73.0 (Firmware 1.0.1)
All the examples appear to be working and the image example is no longer crashing.

Unit test all pass on my updated Flipper running 1.0.1 firmware.

Changes:
- Update SDK bindings to 73.0
- Switch to `unstable-2024-09-10` toolchain (i.e. 1.83.0-nightly)
- Remove `toolbox::md5` and `toolbox::sha256` that are no longer exported by the SDK

Fixes: #150