flipperzero-rs/flipperzero-rs #162: Rename binaries in flipperzero-tools

dcoles, 2024-09-13:
It is possible to install the binaries from `flipperzero-tools` using `cargo install`, however the names are too generic and likely confusing outside the context of the project (e.g. [`storage`](https://github.com/flipperzero-rs/flipperzero/blob/main/tools/src/bin/storage.rs)).

They should all be given a distinctive name, for example `flipper-storage` or `flipper-run`.

JarvisCraft, 2024-09-13:
By the way, we may be interested in adding custom [Cargo commands] so that they can be installed via normal flow. This would eventually let us get rid of the dependency on `python`.

[Cargo commands]: https://doc.rust-lang.org/book/ch14-05-extending-cargo.html

JarvisCraft, 2024-09-14:
Thinking about the current project structure, I suggest moving the whole `tools` crate into a separate repository with the following intents:

1. Have separate development environments and versioning schemes for them.
2. Have the crate published to `crates.io` and potentially OS package managers (I would be interested in packaging it for nixpkgs, for instance).
3. Get rid of python scripts (at the moment they are just wrappers for `cargo run ...` of the tools).