flipperzero-rs/flipperzero-rs #189: "Outdated App" error on latest firmware 1.1.2

CrowhopTech, 2024-11-13:
I tried compiling the basic template app and when running it on my Flipper that I just updated the firmware on, I get the same error as in #150 "Error: Outdated App". It just upgraded to FW version `1.1.2` (commit `54fd2819`) released yesterday (Nov 11 2024).

Hopefully an easy fix but wanted to raise an issue so you were aware. Thanks in advance!

dcoles, 2024-11-13:
Hi @CrowhopTech. Thanks for opening an issue. The current git version supports the 1.1.2 firmware (#187), but we haven't tagged the 0.13.0 release (#188) quite yet.

Just waiting to see if there is any other PRs that can make it into this release, otherwise I'll tag 0.13.0 in the next day or so.

CrowhopTech, 2024-11-24:
Thanks so much! Eagerly awaiting the release 😺

CrowhopTech, 2024-11-24:
I just confirmed the sample app ran with the latest 0.13.0 release 🥳 will close this as it appears good.

Thanks for your quick response!

closed 2024-11-24 by CrowhopTech

dcoles, 2024-11-25:
Thanks for the confirmation! Glad to hear it's working.