flipperzero-rs/flipperzero-rs #193: Missing(?) static for BLE in `sys` crate

Sebbl0508, 2024-12-23:
Hello,

i'm trying to write app using BLE for the flipper. I'm looking at some C code for reference and see they're calling `bt_profile_start` like this:
```c
FuriHalBleProfileBase* ble_hid_profile = bt_profile_start(bt, ble_profile_hid, NULL);
```

`ble_profile_hid` is a `FuriHalBleProfileBase*` defined in `lib/ble_profile/extra_profiles/hid_profile.[c/h]` like this:
```c
static const FuriHalBleProfileTemplate profile_callbacks = {
    .start = ble_profile_hid_start,
    .stop = ble_profile_hid_stop,
    .get_gap_config = ble_profile_hid_get_config,
};

const FuriHalBleProfileTemplate* ble_profile_hid = &profile_callbacks;
```

I haven't found `ble_profile_hid` in the `flipperzero_sys` crate. I don't know if it's missing or if i just haven't found it.

I tried constructing the `profile_callbacks` myself but the functions used in that struct instance are all `static` C functions in `lib/ble_profile/extra_profiles/hid_profile.c`.

Am i on the wrong track or is there something i can do to make this work?

dcoles, 2024-12-24:
Hi @Sebbl0508,

So it looks like [`ble_profile_hid`](https://sourcegraph.com/github.com/flipperdevices/flipperzero-firmware@1.1.2/-/blob/targets/f7/api_symbols.csv?L3726) is not exported by the Flipper SDK (note the `-` in the second column). 

I see that `ble_profile_hid` is being used in the BadUSB app, so you might want to open a ticket on flipperdevices/flipperzero-firmware asking if that symbol can be publicly exported.

Sebbl0508, 2024-12-24:
Hi,

thanks for your answer. I'll have a look^^

dcoles, 2024-12-27:
Closing since this is something that would have to be addressed in the upstream firmware.

closed 2024-12-27 by dcoles