flipperzero-rs/flipperzero-rs #205: Update to Rust Edition 2024

loftyinclination, 2025-01-25:
This is landing soon, and it'd maybe be nice to update. Not sure what effect updating would have on the msrv, but the 2024 edition does come with a few nice features for embedded code and FFI (in particular, declaring [functions in `unsafe extern` blocks as `safe`](https://dev-doc.rust-lang.org/edition-guide/rust-2024/unsafe-extern.html)).

I've tried doing this update locally and found a few minor issues that were easily fixed; 
* [`gen` is now a reserved keyword](https://dev-doc.rust-lang.org/edition-guide/rust-2024/gen-keyword.html), and so a few of the bindings need escaping.
* [linker attributes are now unsafe and need to be marked as such](https://dev-doc.rust-lang.org/edition-guide/rust-2024/unsafe-attributes.html). Think this is probably safe to do?
* extern blocks are now required to be marked as unsafe, and a number of the blocks in the v0.13 bindings weren't
* [the `unsafe_op_in_unsafe_block` lint is now a warning by default](https://dev-doc.rust-lang.org/edition-guide/rust-2024/unsafe-op-in-unsafe-fn.html); this is pretty easily solved, but may just warrant the lint being disabled.

Would be interested to hear your thoughts on updating to this new edition. I've got a local branch that I could raise a draft PR for if that would be helpful.

loftyinclination, 2025-01-25:
Not sure if this is a local issue, but after changing edition to 2024, a number of the string conversion functions (panic, logging, and some furi string methods) fail to compile due to expecting a u8 pointer but instead receiving i8 pointers.

`error[E0308]: mismatched types
    --> flipperzero/examples/i2c-ds3231.rs:79:9
     |
79   |         error!("DS3231 is not connected and ready");
     |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
     |         |
     |         expected `*const u8`, found `*const i8`
     |         arguments to this function are incorrect
     |
     = note: expected raw pointer `*const u8`
                found raw pointer `*const i8`
note: function defined here
    --> /home/lofty/Documents/Creative/Programming/Rust/flipper-zero-api/crates/sys/src/bindings.rs:1968:12
     |
1968 |     pub fn furi_log_print_format(
     |            ^^^^^^^^^^^^^^^^^^^^^
     = note: this error originates in the macro `$crate::log` which comes from the expansion of the macro `error` (in Nightly builds, run with -Z macro-backtrace for more info)`

this seems to be caused by the return value of `CStr::as_ptr`, `core::ffi::c_char` being platform specific. switching to use `*const c_char` instead of `*const i8` or `*const u8` in the logging methods seems to fix this.

dcoles, 2025-01-26:
Hi @loftyinclination,

Thanks for opening this issue, so we can track what would be needed to switch to 2024 edition.

As you point out, it looks like the currently generated aren't compatible with the new edition. These are generated by bindgen, so we probably need to first wait for bindgen support for the 2024 edition.

- https://github.com/rust-lang/rust-bindgen/issues/2901
- https://github.com/rust-lang/rust-bindgen/issues/3093

It looks like the [unsafe attributes](https://dev-doc.rust-lang.org/edition-guide/rust-2024/unsafe-attributes.html) are supported in all editions of Rust from 1.82.0 forward. We currently have the Rust version set to 1.81.0, but I have no problem with us updating that to a newer version (and more recent nightly than 2024-09-10).

I think a good first step would be to enable the `rust-2024-compatibility` [lint group](https://doc.rust-lang.org/rustc/lints/groups.html) which would allow tracking down the various compatibility issues.

```toml
[lints.rust]
rust_2024_compatibility = "warn"
```

We probably won't get them fixed before I cut the [0.14.0 release](https://github.com/flipperzero-rs/flipperzero/issues/203), but I think it would be a good thing to work on once that's out the door.