flipperzero-rs/flipperzero-rs #208: Flipper fails to load app with "Error: Missing Imports" despite API version matching

AverseABFun, 2025-01-27:
For some reason, for my app, every time I attempt to load it on my flipper I get the same error saying that it's missing imports. This in general means that the flipper zero firmware is out-of-date, however the firmware is up to date and has the same API version.

dcoles, 2025-01-28:
Hi @AverseABFun,

Sorry to hear that your app isn't working. I'll need a bit more information from you to be able to track down the issue.

What is the version of the `flipperzero` crates you have in your `Cargo.toml`?

Can I also get you to paste the outputs of the serial log after running your application with `Settings > System > Log Level` set to `Debug` and `Settings > System > Debug` set to `ON`. You should be able to see the output by running the `log` command in the [serial console](https://lab.flipper.net/cli).

I'm most likely looking for something before an `[E][Elf] Error relocating section '…'` message.

Alternatively if you could attach a `.fap` file that causes the issue, that would be extremely helpful for investigating.

AverseABFun, 2025-01-29:
Sorry for the delay, testing now.

dcoles, 2025-02-19:
Hi @AverseABFun. Did you have any luck collecting the logs? If this is no longer a problem, I'll close out this issue in a few days.

dcoles, 2025-04-21:
Closing due to no response.

closed 2025-04-21 by dcoles